riscv-software-src/riscv-isa-sim #1803: Default `pmpcfg0 = 0x000000000000001f` and `pmpaddr0 = 0x003fffffffffffff`.

Donkey-Hao, 2024-09-12:
If we do nothing but read the `pmpcfg{0,2,4,6,8,10,12,14}` and `pmpaddr[i]`, we will see the `pmpcfg0 = 0x000000000000001f` and `pmpaddr0 = 0x003fffffffffffff`. The rest registers are all `0x0000000000000000`.
[image]

[Here](https://github.com/user-attachments/files/16980302/spike_pmp.txt) is a test program.

aswaterman, 2024-09-12:
There’s no bug here. The reset value of the PMP registers is platform-defined. Spike does this for backwards compatibility with old software that wasn’t PMP-aware. If you don’t like these values, your software can overwrite them.

Donkey-Hao, 2024-09-13:
Hi @aswaterman, thanks for your replying. I agree with you, it's not a bug. I was confused by this, but now I am clear. Thanks. 
I have read the chapter in the latest version manual **3.7.1. Physical Memory Protection CSRs**. But I am not totally understood PMP yet. What if the reset value of the PMP registers in a CPU is not all zero when the CPU starts, such as: `pmpcf{i} = 0x00`, but `pmpaddr{j} != 0x00`, is it a bug?

TheSaltyLiquorice, 2024-09-16:
> Hi @aswaterman, thanks for your replying. I agree with you, it's not a bug. I was confused by this, but now I am clear. Thanks.
> I have read the chapter in the latest version manual **3.7.1. [...]

If the reset of the csr prevents M mode code from running, that would be fatal. For example, if pmpaddr0 was all ones, address matching mode tor, the lock bit is set, and X permissions are missing that would prevent the cpu from doing anything useful forever.

On the other hand, as long as the L bit isn't set, machine mode software can do as it pleases with the pmp configuration.

aswaterman, 2024-09-16:
Right.  The platform can specify a nonzero reset value for some PMP registers, and it's the responsibility of the platform to not specify a nonzero reset value that would prevent the processor from functioning (e.g. @TheSaltyLiquorice's example).

Donkey-Hao, 2024-09-21:
Thank you all for clarifying my doubts.

closed 2024-09-21 by Donkey-Hao