riscv-software-src/riscv-isa-sim #1806: Using RISCV FESVR for lockstep verif in C

HakamAtassi, 2024-09-14:
Hello!

I'm currently in the process of bringing up a testbench for a RISCV core I'm working on. Based on what I was able to find online, it seems that there is a way to use the `fesvr` directory to create a spike instance in C and pass it instructions or something to that effect. However, its unclear to me how I should go about setting that up. I've used spike before, but the provided setup process only generates a spike bin in /usr/bin/spike. How do I integrate spike into a C based verification framework? Any guidance or a link to a similar github issue would be great. Thanks.

HakamAtassi, 2024-09-14:
Alright so using https://github.com/ucb-bar/testchipip/blob/master/src/main/resources/testchipip/csrc/cospike_impl.cc as a template has given me an idea of what to do to use spike as an object in C, but I'm still not entirely clear on how build the spike library to actually run this code. I believe I see all the necessary code in `~/riscv-gnu-toolchain/spike/riscv/`, but how do I build/use it?

jerryz123, 2024-09-14:
Generally RTL simulator compilers will allow you to pass in C files alongside your RTL, along with additional arguments to the linker. You should pass in a "cospike_impl.cc-like" file to your RTL simulator, along with `-lriscv` to the linker.

HakamAtassi, 2024-09-14:
Hey, 
Thanks for the response! I was actually more looking for how to actually sun a simulation in pure C atm. I'll worry about crossing the integration bridge when I get to it. I ended up scrapping together a bash script from the make file 

https://github.com/SpinalHDL/rvls/blob/main/Makefile

and I seem to be up and running.

closed 2024-09-14 by HakamAtassi