riscv-software-src/riscv-isa-sim #1825: bf16_mulAdd raises underflow flag when it should not

PingTakPeterTang, 2024-09-30:
@aswaterman 

A = 0x0fb1 = 16'b0_00011111_0110001 = 1.0110001 x 2 ^(-96)
B = 0x0044 = 16'b0_00000000_1000100 = 0.1000100 x 2 ^(-126)
C = 0x8080 = 16'b1_00000001_0000000 = 1.0000000 x 2 ^(-126)
with FRM = RNE, and "tiny" defined as "after rounding"

The product is extremely small, so that |A*B + C| is just smaller than 2^(-126) by an extremely small amount. With RNE and "rounding with unbounded exponent" the value would be normal. But the code reported underflow.

The current code computes A*B in fp64, and performs (A*B) + C in fp64 in RTO, followed by fp64_to_fp32 in RTO, followed by fp32_to_bf16 in prevalent rounding. In this case, the fp64_to_fp32 under RTO reported the underflow (as it should) but not appropriate for the prevalent RNE mode. Looks like we must take all flags raised by fp64_to_fp32 except, and never, accepting the underflow flag. (?) A truly deserved underflow will be generated by the final fp32_to_bf16 conversion. (just a thought, not rigorously verified)

aswaterman, 2024-09-30:
I agree this looks like a bug, but I don't think the proposed solution is correct, because some invocations of `fp64_to_fp32` will raise the underflow exception legitimately (and the subsequent `fp32_to_bf16` will fail to do so).

I think we just need to rewrite `f64_to_bf16` natively to avoid this bug.

liangkaiwang, 2024-09-30:
I think the proposed solution should work. FP64 to FP32 with RTO mode does not produce carry to change the result into the normal range in general. So deferring the underflow operation to FP32->BF16 should not produce incorrect underflow result.

PR #1826 (2024-10-01): Fix f64_to_bf16 raising underflow when it shouldn't
Resolves #1825
commit c3f324ff (2024-10-01): Fix f64_to_bf16 raising underflow when it shouldn't

Resolves #1825
closed 2024-10-01 by aswaterman