riscv-software-src/riscv-isa-sim #1833: Running into illegal instruction traps for floating point instructions

christian-herber-nxp, 2024-10-14:
I am trying to run a floating point workload. I noticed instret counts being really high when using F/D compared to Zfinx/Zdinx.
I am running with the following ISA string:
spike --isa=rv32imafdc_zicsr_zicntr
and I am using the pk.

It seems whenever I hit a F/D instructions, it traps:

```
(spike) 
core   0: 0x00010b54 (0x000027a2) c.fldsp fa5, 8(sp)
core   0: exception trap_illegal_instruction, epc 0x00010b54
core   0:           tval 0x000027a2
(spike) 
core   0: >>>>  trap_vector
core   0: 0x80000004 (0x34011173) csrrw   sp, mscratch, sp
(spike) until pc 0 10b5c
(spike) 
core   0: 0x00010b5c (0x4afdf7c3) fmadd.d fa5, fs11, fa5, fs1
core   0: exception trap_illegal_instruction, epc 0x00010b5c
core   0:           tval 0x4afdf7c3
```

Am I using this wrong? Doesn't seem to be the intended behavior.

When I check misa, I get:

```
(spike) reg 0 misa
0x40141105
```

which is `01000000000101000001000100000101`, which indicates F and D are not supported.

aswaterman, 2024-10-14:
My best guess is that you are using a PK built for non-FP.  Reconfigure with the right `--with-arch` string and you should be good.

christian-herber-nxp, 2024-10-14:
> My best guess is that you are using a PK built for non-FP. Reconfigure with the right `--with-arch` string and you should be good.

works. thanks.

closed 2024-10-14 by christian-herber-nxp

aswaterman, 2024-10-14:
You're welcome!