riscv-software-src/riscv-isa-sim #1844: vid.v interaction with vstart

rbuchner-aril, 2024-10-26:
From the vector spec, section 15.9, `The vid.v instruction writes each element’s index to the destination vector register group, from 0 to vl-1.` 
 
 In Spike, the main loop in `riscv/insns/vid_v.h`:
 ```
 for (reg_t i = P.VU.vstart->read() ; i < P.VU.vl->read(); ++i) {
 ```

From the spec, I would expect this loop to read:
```
for (reg_t i = 0 ; i < P.VU.vl->read(); ++i) {
```

However, I would also think that it should take an illegal instruction then for when `vstart != 0`  if we made this change as to not silently ignore vstart.

 Anyways, just hoping to get clarification on this instruction, I can put together a MR if you think a change would be necessary.

aswaterman, 2024-10-28:
As the spec also says, "All vector instructions are defined to begin execution with the element number given in the vstart CSR".

The spec's description of `vid.v` discusses what happens for the elements that are active and doesn't repeat the boilerplate about the so-called prestart elements.

rbuchner-aril, 2024-10-28:
Thanks for following up. I got a little tripped up since `vl` was explicitly mentioned but not `vstart`. I do agree that Spike's interpretation is valid.

As always, thanks for the timely response.

closed 2024-10-28 by rbuchner-aril

aswaterman, 2024-10-28:
Yeah, it's not crystal clear, but at least it isn't broken.

You're welcome!