riscv-software-src/riscv-isa-sim #1846: Getting illegal instruction traps when trying to use Zfinx/Zdinx

christian-herber-nxp, 2024-10-29:
Hi recompiled spike from the master branch. 
I still have an old binary I compiled earlier. 
With the old one, no problems.
With the new one, I get an illegal instruction trap whenever an f... instruction is seen. 
This is a minimal example that doesn't run right now:

#include <stdio.h>

```
int main() {
	volatile float b = 7.0;
	printf("Hello World %f\n", b);
	return 0;
}
```
Running leads to this:

```
spike -d --isa=rv32imac_zba_zbb_zbs_zicsr_zicond_zicntr_zce_zdinx --misaligned pk -p a.out
(spike) until pc 0 00010198
bbl loader
(spike) 
core   0: 0x00010198 (0x42050553) fcvt.d.s a0, a0
core   0: exception trap_illegal_instruction, epc 0x00010198
core   0:           tval 0x42050553
(spike)
```

aswaterman, 2024-10-29:
I don't have time to debug this one, but if anyone does, it sounds like it's worth looking into.

demin-han, 2024-10-31:
Does sstatus.FS also control Zfinx?
if yes:
  const bool has_fs = (proc->extension_enabled('S') || proc->extension_enabled('F')) 
if no:
  #define require_fs          require(STATE.sstatus->enabled(SSTATUS_FS) || proc->extension_enabled(EXT_ZFINX))

The fs field is masked when enable Zfinx.

reg_t base_status_csr_t::compute_sstatus_write_mask() const noexcept {
  // If a configuration has FS bits, they will always be accessible no
  // matter the state of misa.
  _const bool has_fs = (proc->extension_enabled('S') || proc->extension_enabled('F')) && !proc->extension_enabled(EXT_ZFINX);_
  // Implementations w/o V may still have mstatus.vs,
  const bool has_vs = proc->any_vector_extensions();
  return 0
    | (proc->extension_enabled('S') ? (SSTATUS_SIE | SSTATUS_SPIE | SSTATUS_SPP) : 0)
    | (has_page ? (SSTATUS_SUM | SSTATUS_MXR) : 0)
    | (has_fs ? SSTATUS_FS : 0)
    | (proc->any_custom_extensions() ? SSTATUS_XS : 0)
    | (has_vs ? SSTATUS_VS : 0)
    | (proc->extension_enabled(EXT_ZICFILP) ? SSTATUS_SPELP : 0)
    | (proc->extension_enabled(EXT_SSDBLTRP) ? SSTATUS_SDT : 0)
    ;
}

But requrie_fs need SSTATUS_FS.
  #define require_fs          require(STATE.sstatus->enabled(SSTATUS_FS))

aswaterman, 2024-10-31:
FS should be hardwired to 0 when Zfinx is implemented.  Spike gets that right, but then still checks for FS being nonzero elsewhere, which is a bug.  I'll open a PR to fix this.

demin-han, 2024-10-31:
> FS should be hardwired to 0 when Zfinx is implemented. Spike gets that right, but then still checks for FS being nonzero elsewhere, which is a bug. I'll open a PR to fix this.
OK.
the no branch above mentioned need more code to change because of fcsr update relies on dirty().

aswaterman, 2024-10-31:
@demin-han Thanks for the reminder; that probably saved me 15 minutes.

closed 2024-10-31 by aswaterman