riscv-software-src/riscv-isa-sim #1856: Problems with the performance of the lhu command on spike

90ICEDA, 2024-11-13:
Type of issue: bug report
Impact: unknown
Development Phase: proposal
Other information：

Hi, I noticed a difference between Rocket，boom and spike implementations, when executing the lhu x8,-7(x21) instruction, spike will execute normally while Rocket，boom will report an error about not being aligned, is this a stricter measure taken by Rocket，boom in handling alignment errors? or maybe spike didnot consider this problem？

I hope to get your reply, please correct me if my thinking is wrong, thanks!

jerryz123, 2024-11-13:
Neither rocket nor boom support misaligned load/stores. This is allowed by the specification.

On those cores, misaligned loads/stores would typically trap to a M-mode handler that emulates the access.

90ICEDA, 2024-11-14:
oh, thank you for your reply!

closed 2024-11-14 by aswaterman