riscv-software-src/riscv-isa-sim #1878: dcsr.debugver field default value

Madman-Hugo, 2024-12-17:
Spike’s dcsr.xdebugver is 1 while spec is 0/4/15
[image]

aswaterman, 2024-12-17:
It appears that @YenHaoChen accidentally introduced this bug in ec292be4fdac9685c9d6f2f209b73743c2e573f7.

Prior to that change, Spike defined this field to occupy bit positions 31:30, and so inserting a 1 into that position was equivalent to encoding the value 4 in bit positions 31:28.  After the change, the field was redefined to occupy bit positions 31:28, but the value was not changed from 1 to 4.

aswaterman, 2024-12-17:
Fixed, thanks for pointing this out.

PR #1880 (2024-12-17): Set dcsr.xdebugver to 4, as it ought to be
See discusson on #1878.  This restores the behavior prior to ec292be4fdac9685c9d6f2f209b73743c2e573f7, which inadvertently changed the value to 1.

Resolves #1878
commit 0fea35dd (2024-12-17): Set dcsr.xdebugver to 4, as it ought to be

See discusson on #1878.  This restores the behavior prior to
ec292be4fdac9685c9d6f2f209b73743c2e573f7, which inadvertently changed
the value to 1.

Resolves #1878
closed 2024-12-17 by aswaterman