riscv-software-src/riscv-isa-sim #1879: Can spike-dasm disassemble specified extensions through the --isa option?

ha0lyu, 2024-12-17:
When I run spike-dasm to disassemble instructions, it disassembled vector extension. When I specify the extensions, it ignored. It seems that this tool could disassemble any instruction.

[image]

I have read the code of [spike-dasm.cc](https://github.com/riscv-software-src/riscv-isa-sim/blob/master/spike_dasm/spike-dasm.cc), it supports the DEFAULT_ISA, which is `rv64imafdc_zicntr_zihpm`.  In the code, spike-dasm will parse `isa` option, but it fails to disassemble instructions of specified extensions. **Is this normal?**

**version**: latest(a2dcf1f), 1.1.1-dev

aswaterman, 2024-12-17:
Yes, this is the intended behavior.  We always try to disassemble an instruction even if it corresponds to an extension that isn't enabled.

For this program, the only purpose of the `--isa` argument is to disambiguate conflicting cases, where a given opcode corresponds to a different instruction depending on which extensions are enabled.  For example:

```
$ echo "DASM(0xb842)"|spike-dasm --isa=rv64imac_zcmp_zcmt
cm.push {ra}, -16
$ echo "DASM(0xb842)"|spike-dasm --isa=rv64imafdc
c.fsdsp fa6, 48(sp)
```

ha0lyu, 2024-12-18:
Hi @aswaterman,
Thanks for your reply. I got it.
I have another question about specified extensions disassemble.
> For this program, the only purpose of the --isa argument is to disambiguate conflicting cases

Now, I want to change the source code to disassemble specified extensions by the --isa argument or another argument. I have checked the code: `disasm.cc` and `isa_parser.h`. The constructor function of class `disassembler_t` calls `add_instructions` function firstly. The `add_instructions` adds instructions according to `*isa`, then it adds instructions of each `isa->extension_enable`. Actually, this tool can not disassemble all instructions, it only supports rv32 and rv64 instructions. Here is my question: how can I change the code to support specified XLEN and extensions, such as rv32imcb, rv64gc, rv32gc, etc.

I need a tool to disassemble instructions. Since Spike is considered the golden model, I believe the best approach is to modify spike-dasm. As someone who is highly knowledgeable about Spike, could you please provide me with some guidance?

aswaterman, 2024-12-18:
I’m not sure I follow exactly what you’re trying to do. Could you give some user stories (really just examples of input, command-line options, and desired output) that would illustrate your goal?

ha0lyu, 2024-12-18:
For example, I run `spike-dasm --isa=rv32i`, spike-dasm will only disassemble `rv32i` instructions.

1. Such as `bits= 0x3256  # c.fldsp ft4, 368(sp)`, this is a C extension instruction, spike-dasm will not disassemble it.  
2. Such as `bits = 0x0182b283  # ld   t0, 24(t0)`, this is a RV64I instruction, spike-dasm will not disassemble it.  

In another case, I run `spike-dasm --isa=rv64g`, it will disassemble RV32g instructions, but will not disassemble C extension instructions. 
I am not sure am I understood, please let me know if you are confused. Thanks.

aswaterman, 2024-12-18:
See #1881

ha0lyu, 2024-12-18:
I have tested it, it works. Nice job! 
Thank you very much!

aswaterman, 2024-12-18:
You're welcome!

PR #1881 (2024-12-20): Add spike-dasm --strict option 
Prevents disassembling instructions that are not enabled by the `--isa` setting.

Resolves #1879