riscv-software-src/riscv-isa-sim #1890: exception trap_store_page_fault

soccor-crypto, 2025-01-10:
When the CPU tries to read and write the index address form a 4k page table, the latest version of spike encounters the following error, 
core  15: exception trap_store_page_fault, epc 0xfffffff00006cd8e
core  15:           tval 0x0000000e56ab9b40
while previous versions do not,
core  15: 0xfffffff00006cd8e (0x00f2b027) fsd     fa5, 0(t0)
core  15: mmu stage1 walk vaddr 0xe56ab9b40
core  15: mmu stage1 walk level 2 index 57 pte addr 0x800001c8 pte 0x2000e8c1(next base 0x8003a000)
core  15: mmu stage1 walk level 1 index 181 pte addr 0x8003a5a8 pte 0x21d2d8c1(next base 0x874b6000)
core  15: mmu stage1 walk level 0 index 185 pte addr 0x874b65c8 pte 0x660d98cf(next base 0x198366000)
core  15: mmu vaddr 0xe56ab9b40 get paddr 0x198366000
core  15: 1 0xfffffff00006cd8e (0x00f2b027) mem 0x0000000e56ab9b40 0x4016215079c605c6
But the old version doesn't support the cbo Instruction and the new version supports the cbo Instruction so I prefer to use the new version. This is a weird question.

aswaterman, 2025-01-10:
> This is a weird question.

I don't think you actually asked a question.

closed 2025-01-10 by aswaterman

soccor-crypto, 2025-01-11:
This is obviously a bug introduced in the new version, the new version does not get the physical address correctly, the new version should not have this problem? I did a lot of testing, the old version can get the physical address just like the above log, but the new version can't do it.

soccor-crypto, 2025-01-11:
This is obviously a bug introduced in the new version, the new version does not get the physical address correctly, the new version should not have this problem? I did a lot of testing, the old version can get the physical address just like the above log, but the new version can't do it.It seems that the new version does not start the MMU.