riscv-software-src/riscv-isa-sim #1894: vmulh* and ZVE64* question

aprnath, 2025-01-15:
Does Spike support this statement from the spec: 

```All Zve* extensions support all vector integer instructions (Section Section 31.11), except that the vmulh integer multiply variants that return the high word of the product (vmulh.vv, vmulh.vx, vmulhu.vv, vmulhu.vx, vmulhsu.vv, vmulhsu.vx) are not included for EEW=64 in Zve64*.```

When using zvl128b_zve64f, Spike does not seem to tale an illegal_instruction exception for vmulh.vv. Is there something specific that is needed to enable this behavior?

aswaterman, 2025-01-15:
This is not technically a violation of the ISA, as the reserved behavior of unimplemented opcodes does not mandate an illegal-instruction exception.  Nevertheless, it is an error of intent.  I agree that Spike should be changed to raise an illegal-instruction exception for these instructions for SEW=64 when the full V extension is not provided.  If you want to make a PR that does so, go for it.

Derrick0415, 2025-03-06:
hi, sorry to bother. I'm trying to run a zve64x program (compiled using clang).
first i encountered a isa error `unsupported extension: zve64x`, so i switch it to `rv64gcv`.
then i keep get this illegal instruction error during execution as pic shows below.
I double check the source is at master branch and up to date, i cant understand whats wrong
Can you kindly give me some tips about how to make it work?

[image]