riscv-software-src/riscv-isa-sim #1898: Wrong senvcfg check for systems without S-mode support

mimiqdev, 2025-01-21:
In Spike's `require_envcfg` macro, it checks senvcfg even when S-mode is not supported:

```cpp
#define require_envcfg(field) \
  do { \
    if (((STATE.prv != PRV_M) && (m##field == 0)) || \
        ((STATE.prv == PRV_U && !STATE.v) && (s##field == 0))) \
      throw trap_illegal_instruction(insn.bits()); \
    else if (STATE.v && ((h##field == 0) || \
                        ((STATE.prv == PRV_U) && (s##field == 0)))) \
      throw trap_virtual_instruction(insn.bits()); \
  } while (0);
```
This is problematic for systems that don't implement S-mode. The current implementation would incorrectly check s##field (like senvcfg.CBIE for CBO.FLUSH instruction) even when S-mode is not supported. 
The discussion in the CMO spec repo https://github.com/riscv/riscv-CMOs/issues/71  

Possible fixes:
```cpp
#define require_envcfg(field) \
  do { \
    if ((STATE.prv != PRV_M) && (m##field == 0)) \
      throw trap_illegal_instruction(insn.bits()); \
    if (p->extension_enabled_const('S') && (STATE.prv == PRV_U && !STATE.v) && (s##field == 0)) \
      throw trap_illegal_instruction(insn.bits()); \
    else if (STATE.v && ((h##field == 0) || \
                        (p->extension_enabled_const('S') && (STATE.prv == PRV_U) && (s##field == 0)))) \
      throw trap_virtual_instruction(insn.bits()); \
  } while (0);
```

aswaterman, 2025-01-21:
Your proposed fix looks correct to me.  Would you mind issuing a pull request?

mimiqdev, 2025-01-21:
Thanks for the reply, I'll create a PR.