riscv-software-src/riscv-isa-sim #1900: Flush TLB entries on multiple events

RohanAhmed1, 2025-01-21:
I see the spike code and found out that, spike flushes the TLB in those events,

- write of MSTATUS, SATP, PMP and PMPCFG
- Handle Trap
- XRET 

but I didn't found this cases of flushing the TLB in riscv spec. except in the case of sfence.vma instruction. So, any specific reason, why did we flushing the TLB's in above cases.

aswaterman, 2025-01-21:
Implementations are always free to flush the TLB more often than required by the ISA.  For Spike, doing so simplifies the simulator, and in some cases improves the simulator's performance.  For example, flushing the TLB at privilege-mode changes means that TLB accesses don't need to explicitly perform permissions checks.  This is not a tradeoff you'd made in a hardware implementation, but it's beneficial in this context.

RohanAhmed1, 2025-01-22:
Right, Thanks. if we didn't have to flush the TLBs in these events. Then what type of stuff/permission check, we have to to implement in TLB just like hardware implementation. Can you guide, where can I found those details.

aswaterman, 2025-01-24:
This assumption is pretty heavily baked in; it is not going to be a straightforward change.  At minimum, you'd need to augment the TLB entries with permissions info for each privilege mode, then check that information on each load/store/fetch.  But there are no doubt a bunch of extra, minor cases.

If your goal is to model the performance of a different TLB, then I'd augment Spike with a separate TLB timing model (i.e., separate the timing model from the functional model altogether).  If your goal is functional verification, then it'll be a challenge.