riscv-software-src/riscv-isa-sim #1906: Correctness of Handling an 8-Byte Instruction Spanning Two PMP Regions

dmpetrow, 2025-01-27:
I am testing PMP. On the processor under test, when executing an 8-byte instruction that spans two regions, an exception occurs. I believe this behavior is correct according to the ISA. Here's the relevant quote:

_The matching PMP entry must match all bytes of an access, or the access fails, irrespective of the L, R, W, and X bits. For example, if a PMP entry is configured to match the four-byte range 0xC–0xF, then an 8-byte access to the range 0x8–0xF will fail, assuming that PMP entry is the highest-priority entry that matches those addresses._

On Spike, no exception is triggered. Am I correct in understanding that if an instruction is valid in terms of RWX permissions across two regions, Spike does not raise an exception?

aswaterman, 2025-01-27:
This is a corner case where both implementations are correct.  Because the memory access is misaligned, the implementation has the freedom to execute it as a single memory access or crack it into multiple smaller ones.  In the former case, the PMP exception is raised because the access spans a region boundary.  In the latter case, each individual access fits within a region, therefore no exception is raised.

If this were, say, a naturally aligned 8-byte access on an RV64 machine that spanned a region boundary, then the exception would be mandatory.

dmpetrow, 2025-01-28:
OK. Will you accept a pull request that introduces a command-line key to trigger an exception for processors with hardware misaligned access so that the behavior in such cases is consistent? I think this could be useful for other developers.

aswaterman, 2025-01-29:
I can't say I'm wild about the idea... Spike isn't meant to model all possible RISC-V systems, so a smattering of command-line instructions to support a fairly random part of the design space is of restricted utility.  It might be better to change it on a branch for your purposes.

closed 2025-01-30 by dmpetrow