riscv-software-src/riscv-isa-sim #1917: Question about restriction on a mtinst/htinst update when written by a CSR operation

viktoryou, 2025-02-10:
From Privilege spec, 
> mtinst is a WARL register that need only be able to hold the values that the implementation may automatically write to it on a trap.

In a CSR write operation, the spike would allow any value to be updated in mtinst/htinst, while only value in Table 39(special pseudoinstruction values for guest-page faults) is supported to automatically write at present. Should the value of *tinst be restricted according to the transformed Instruction or pseudoinstruction that actually implemented?

aswaterman, 2025-02-10:
We have some flexibility here.  The statement in the spec specifies a lower bound on which values need be representable, not an upper bound.  Spike normally doesn't aggressively take advantage of this sort of property, because it isn't advantageous for a software-based simulator to do so.  (Similarly, we don't trim _x_`tval` to the maximum supported address size.)  I just assume leave it alone.

viktoryou, 2025-02-11:
From my understanding, the restriction still needs to be applied to implementation by the WARL rule: when*tinst CSR could automatically be written by 0/0x3000/0x3020(no other value supported) in RV64 system on a trap, then only 0/0x3000/0x3020 should be legally read. The Spike keeps its flexibility and the implementation observes the rule, right?
Another puzzling thing is, what would be the use case of updating *tinst by a CSR write operation? Is it allowed to ignore the write by CSR operation for simplicity?

aswaterman, 2025-02-11:
The implication is that the set of legal values is greater than the set of values the machine might automatically write.  There isn't actually a contradiction here.

closed 2025-02-11 by viktoryou