riscv-software-src/riscv-isa-sim #1919: *tval took the transformed address of PA/GPA when satp is bare and pointer masking is enabled

viktoryou, 2025-02-12:
From the spec, 
> pointer masking is applied for hardware writes to a CSR (e.g., when the hardware writes the transformed address to stval when taking an exception)

On a trap delivery, *tval took the transformed address of GPA when vsatp is bare in a virtualization environment and pointer masking is enabled. As the mtval/stval will contain the faulting (guest)virtual address, should reported transformed address be in a form of virtual address(sign-extension) instead of (guest)physical address(zero extension, the way spike adopted)?
Could @YenHaoChen or @aswaterman or @martinmaas or someone help to check this point?

aswaterman, 2025-02-13:
In general, the address written to xtval must be exactly the same as the transformed address, if the transformed address was valid; or, if the transformed address was invalid, the value written must be some invalid address (possibly exactly the same as the transformed address).

In this context, an address is "valid" if (a) first-stage address translation is enabled and the address is canonically sign-extended, or (b) first-stage address translation is disabled and the address is a valid physical address.

It's not clear whether the address in your example is valid: e.g. the trap could've been caused for other reasons, e.g. insufficient permissions in second-stage translation, in which case the address would still be considered valid.  So it's not clear what the legal xtval values might be. 
 But if the address is invalid, then any invalid address could be legally written to xtval.

viktoryou, 2025-02-14:
The original question is due to ambiguous description about the ignore transformation applied in the cases where the
active satp register’s MODE field is Bare. Seems https://github.com/riscv/riscv-j-extension/issues/50 figured it out. Thanks @aswaterman for the datailed explanation of the rule of updating xtval. I would close this issue.

closed 2025-02-14 by viktoryou

aswaterman, 2025-02-14:
👍 and thanks for cross-linking the other issue for posterity.