riscv-software-src/riscv-isa-sim #1921: vfw* instruction(s) with zve*f extension

aar0n-st0ne, 2025-02-20:
As per the spec for Zve*f extension floating point operands with EEW=64 are not supported:

> All Zve* extensions support all vector permutation instructions (Section Section 31.16), except that Zve32x and Zve64x do not include those with floating-point operands, and Zve64f does not include those with EEW=64 floating-point operands.
> The Zve32x extension depends on the Zicsr extension. The Zve32f and Zve64f extensions depend upon the F extension, and implement all vector floating-point instructions (Section Section 31.13) for floating-point operands with EEW=32. Vector single-width floating-point reduction operations (Section 31.14.3) for EEW=32 are supported.

So, ideally vector floating point widening instructions (eg. vfwadd.vv) should throw an illegal instruction error when SEW=32 as the destination EEW would then be 64. 

When using zvl128b_zve64f, Spike does not seem to take an illegal_instruction exception for this case. Is my understanding wrong or is there something specific that is needed to enable this behavior?

aswaterman, 2025-02-20:
Your understanding is correct.  This is one of a handful of cases where Spike takes liberties with the concept of "reserved", failing to raise an exception where it probably should.  This isn't a violation of the ISA, but it's still undesirable.  If you're willing to prepare a PR to correct this deficiency, I'd be happy to review it.

aar0n-st0ne, 2025-02-24:
Thank you for the clarification!