riscv-software-src/riscv-isa-sim #1922: Supported SEWs for fractional LMUL

aar0n-st0ne, 2025-02-20:
Spec states that

> For a given supported fractional LMUL setting, implementations must support SEW settings between SEWmin and LMUL * ELEN, inclusive.

So, for ELEN=64 and LMUL=1/8 the the SEW settings to be supported are: {8 _(SEWmin)_, 8 _(64*1/8)_}.
This calculation is agnostic to the value of VLEN. Let's taken an example where the VLEN is 128. For LMUL=1/8 and same ELEN of 64 the implementation is capable of handling the SEW 8 as well as 16 as VLEN*LMUL = 128/8 = 16 => 1 element of SEW=16 can be processed.

With this exact configuration of VLEN=128 and ELEN=64 when a vset{i}vl{i} instruction is executed with e8 and mf8 config along with avl such that the vl should be set to VLMAX Spike sets the vill bit of vtype register stating this is an illegal configuration.

My understanding is that this given set of SEWs {SEWmin, LMUL * ELEN} is the minimal set which **_must_** be supported by an implementation as VLEN=ELEN is a valid implementation choice. If the VLEN is such that larger SEWs can be supported then an implementation may not throw illegal instruction error. Is my understanding correct or  should an implementation support only these SEWs for a fractional LMUL and throw illegal instruction error or anything else regardless of VLEN?

aswaterman, 2025-02-20:
Put another way, it is valid, but not required, for an implementation to act as though VLEN=ELEN for the purposes of determining which SEW/LMUL combinations are supported.

aar0n-st0ne, 2025-02-24:
Thank you for the clarification!