riscv-software-src/riscv-isa-sim #1924: pointer masking：consider vsstatus.MXR ?

HJM-work, 2025-02-25:
[image]
The description states that the pointer masking does not consider the MXR field of vsstatus, but the implementation code is affected by vsstatus.MXRwhen effective_virt==1?

[image]

aswaterman, 2025-02-25:
Look to the left of what you highlighted.  `sstatus->readvirt(false) | sstatus->readvirt(effective_virt)` means "always apply `sstatus`, and apply `vsstatus` when V=1".

HJM-work, 2025-02-25:
> Look to the left of what you highlighted. `sstatus->readvirt(false) | sstatus->readvirt(effective_virt)` means "always apply `sstatus`, and _additionally_ apply `vsstatus` when V=1". This is what the ISA requires.

Thank you for your reply.

aswaterman, 2025-02-25:
You're welcome!

closed 2025-02-26 by HJM-work