riscv-software-src/riscv-isa-sim #1928: about hgatp register

HJM-work, 2025-03-05:
[image]
“When writing to 'hgatp', if ‘MODE==Bare’, the remaining fields in hgatp must be set to zeros. ” It seems that this logic has not beend handled?
 
[image]

aswaterman, 2025-03-06:
The use of passive voice is confusing, but I think the spec is trying to say that it is software's responsibility to set the other fields to zero (and, implicitly, the behavior is unspecified if software fails to do so).  If my interpretation is correct, then there is no bug in Spike.  @jhauser-us might have an opinion.

jhauser-us, 2025-03-06:
I agree with @aswaterman.  The intention has always been that *software* must set the other fields to zero, same as for `satp`.  The supervisor chapter has this statement for `satp`:
> To select MODE=Bare, software must write zero to the remaining fields of `satp`.

aswaterman, 2025-03-06:
Thanks for the confirmation.  I'll change the wording for `hgatp` to use the active voice, as with `satp`.  Closing this issue, as there's no bug.

closed 2025-03-06 by aswaterman