riscv-software-src/riscv-isa-sim #1935: Spike executed an instruction that does not meet expectations.

ha0lyu, 2025-03-11:
During testing with Spike, I noticed that the instruction at `pc = 0x800005ac` differs from the one executed by Spike.
[image]

As observed, the register has become `hstatus`, which is not the expected value.
[image]

[test.zip](https://github.com/user-attachments/files/19195705/test.zip)

DavidYu360, 2025-03-12:
1. The trace in the post contains hypervisor-related CSR, and maybe the spike is not set for supporting hypervisor. That is why spike raises illegal instruction exception. 
2. I can not find the instruction "csrr sp, hstatus" in the dump or elf file, please have a check if the ELF file in the attachment is the "real" elf file you fed to spike.

heiyuen1999, 2025-03-26:
Check whether there is a store that modifies the highest byte of the address 0x800005ac.

ha0lyu, 2025-04-16:
Apologies for the delayed response. The problem turned out to be with my test case, which contains self-modifying behavior. 
I really appreciate everyone's replies. Thank you!

closed 2025-04-16 by ha0lyu