riscv-software-src/riscv-isa-sim #1940: FOR V=1 does the MXR should be effect the pointer masking apply

xinyuwang-starfive, 2025-03-20:
https://github.com/riscv/riscv-isa-manual/issues/1915 
@aswaterman 
in this issue it say in V=1 the MXR will not effect pointer masking apply, but now the spike is not .

1）When one of vsstatus.MXR or sstatus.MXR is set, of course, V=1 and vsatp.mode is not bare, should pointer masking be applied?
2）And when V=1, vsatp.mode is bare. Should we ignore vsstatus.MXR and only focus on sstatus.MXR for applying pointer masking?

[image]

aswaterman, 2025-03-27:
@ved-rivos Since you were investigating related questions on the mailing list (or at least I think they're related), can you tackle this one?

ved-rivos, 2025-03-27:
The specification states:
     "When MXR is in effect at the effective privilege mode where explicit memory access is performed, pointer masking does not apply."
and further clarifies:
    "Note that this includes cases where page-based virtual memory is not in effect; i.e., although MXR has no effect on permissions checks when page-based virtual memory is not in effect, it is still used in determining whether or not pointer masking should be applied."

To determine whether MXR is in effect, the rule in Table 42 should guide.
1. If the effective privilege mode of the access is VS or VU, then MXR is in effect if either vsstatus.MXR is 1 or sstatus.MXR is 1.
2. If the effective privilege mode of the access is S/HS, then MXR is in effect if sstatus.MXR is 1.

> 1）When one of vsstatus.MXR or sstatus.MXR is set, of course, V=1 and vsatp.mode is not bare, should pointer masking be applied?"

Pointer masking is not applied. The status of vsatp.MODE is immaterial as noted by the pointer masking specification.

> 2）And when V=1, vsatp.mode is bare. Should we ignore vsstatus.MXR and only focus on sstatus.MXR for applying pointer masking?"

For explict accesses to effective privilege mode VS/VU has MXR in effect if sstatus.MXR is 1 or if vsstatus.MXR is 1.

qin996, 2025-03-28:
@ved-rivos hi, ved. where is the `Table 42`? I can't find it in privileged spec, unprivileged spec and pointing mask spec. can you show here? and please help confirm the following questions:
1) if either vsstatus.MXR is 1 or sstatus.MXR is 1,  pointer masking is not applied when h[ls]v, or mprv=1 mpv=1 mpp=0/1, or V=1. is right?
2) if mstatus.MXR=1, should pointer masking be applied in m mode?
3) If the effective privilege mode of the access is U, should pointer masking be applied when sstatus.MXR=1?

thanks

ved-rivos, 2025-03-28:
> if either vsstatus.MXR is 1 or sstatus.MXR is 1, pointer masking is not applied when h[ls]v, or mprv=1 mpv=1 mpp=0/1, or V=1. is right?

Yes.

> if mstatus.MXR=1, should pointer masking be applied in m mode?

`mstatus.MXR` does not affect memory accesses made with effective privilege mode equal to M.

> If the effective privilege mode of the access is U, should pointer masking be applied when sstatus.MXR=1?

No.

xinyuwang-starfive, 2025-04-03:
@ved-rivos  i think now spike if if mstatus.MXR=1 and effective privilege mode equal to M nad MSECCFG is not equal 0，the 
pointer masking is not apply 。 
it not like your say the mstatus.MXR does not affect memory accesses made with effective privilege mode equal to M.

ved-rivos, 2025-04-04:
> it not like your say the mstatus.MXR does not affect memory accesses made with effective privilege mode equal to M.

I see. I think a PR to fix that may be good.