riscv-software-src/riscv-isa-sim #1941: Potentially Incorrect PMP Behavior in Spike: Access Crossing PMP Boundary Allowed

mahendraVamshi, 2025-03-20:
This issue describes PMP behavior when there is an access that crosses a PMP region. Spike seems to allow the access even though, as per spec:

[image]

Problem scenario: I have configured the PMP region to be 80000000 to 8003ffff using NAPOT. 

[image]

Spike command: spike --hartids=0 --log ref.dump --log-commits --priv=msu -m0x80000000:0x4000000,0x1000:0x1000 --disable-dtb --pc=4096 --isa=rv64imafdc_zba_zbb_zbc_zbs_smrnmi_zicntr_zihpm_svnapot --pmpregions=16 --pmpgranularity=64 dut.elf

I expect the simulation to trap when access crosses the PMP boundary but the access seems to be allowed. This can be seen below when accessing 4-byte instruction on PC 0x000000008003fffe. 

[image]

I have attached the elf (dut.elf), disassembly (dut.disass), and the ref.dump in the below zip file

[ref.zip](https://github.com/user-attachments/files/19359769/ref.zip)

Please let me know if this is the expected behavior and why. Thanks a lot.

aswaterman, 2025-03-20:
There's another clause that I believe explains the behavior: https://github.com/riscv/riscv-isa-manual/blob/ee6fbb52887e73506d954d5397ef757165c755de/src/machine.adoc#L3286

Misaligned accesses (including this instruction fetch) can be decomposed into multiple aligned accesses.  If those accesses individually are allowed by PMP, then the entire access is allowed to succeed.

serge0699, 2025-06-09:
Same is #1906.