riscv-software-src/riscv-isa-sim #1943: Misaligned double load and stores are handled differently

dwRchyngqxs, 2025-03-26:
Misaligned double load and stores are handled differently, the load is fine while the store triggers an unhandlable trap (and spikes exit with exit code 0).
The following `main.S` shows the issue (comment and uncomment the fld/fsd lines as you see fit).
```asm
.globl main
.text
main:
addi sp, sp, -9
fld ft0, 0(sp)
fsd ft0, 0(sp)
addi sp, sp, 9
mv a0, zero
jr ra
```
Output is
```
../machine/mtrap.c:22: machine mode: unhandlable trap 7 @ 0x80005b8c
Power off
```
when the misaligned load is encountered and nothing in all other cases.
Compiled with `riscv32-unknown-elf-gcc -o main -s main.S` and run with `spike --isa=rv32g pk main`.

Misaligned load and stores for integers complete successfully (no complaint at all, exit code 0).

aswaterman, 2025-03-27:
Doesn't look like a Spike problem.  Most likely a bug in `pk`'s emulation code, or maybe a problem with how `pk` was configured and compiled.  If you can debug it and submit a PR to `riscv-pk` to fix it, we'd be grateful.

aswaterman, 2025-03-27:
Closing, as the issue is being tracked on the pk repo.

closed 2025-03-27 by aswaterman