riscv-software-src/riscv-isa-sim #1944: mmio & cm.jalt

jmonesti, 2025-03-26:
Hello,

Supporting a fork of Spike for our own purpose, I’m currently integrating it into TLM/SystemC.

In particular, in this environment, Spike does not instanciate its own memories using the well-known mem_t,
but rather a derivative of abstract_device_t -call it tlm_mem_t- that accesses the remote TLM/SystemC memories via function pointers.

This implementation implies that all memory read/write’s go through sim_t::mmio_load() / sim_t::mmio_store()

That’s for the context.

Back to my concern,

When I put the stack in a TLM/SystemC memory, everything’s fine, my firmware is functional.

However, if I put the code in a TLM/SystemC memory, the execution traps such as:

core   0: 0x00020794 (0x0000a0da) cm.jalt 54
core   0: 3 0x00020794 (0xa0da) x1  0x00020796
core   0: 3 exception trap_instruction_access_fault, mepc 0xe7922118, mcause 0x00000001, mstatus 0x00001800, tval 0xe7922118

Considering my jump table, an eagle-eye can pick that the 2 lsb of 0xe792.2118 are correct, the 2 msb being garbage.
Plus, these 2 msb change each every time the test re-runs (ie. smells like uninitialized data)

There began my gdb journey…

cm_jalt.h reads:

`target = MMU.fetch_jump_table<int32_t>(base + (index << 2));`

which is (mmu.h):

```
template<typename T>
  T ALWAYS_INLINE fetch_jump_table(reg_t addr) {
    auto tlb_entry = translate_insn_addr(addr);                                                             [1]
    return from_target(*(target_endian<T>*)(tlb_entry.host_offset + addr));          [2]
  }
```

[1]
translate_insn_addr(addr) calls
fetch_slow_path(addr) which -due to my TLM/SystemC stuff- results in
mmio_fetch(paddr, sizeof fetch_temp, (uint8_t*)&fetch_temp)

fetch_mem being uint16_t, this means that mmio_fetch() returns 2 bytes from the memory.

Unfortunately, [2] will read 4 (rv32) or 8 (rv64) bytes from the ‘fake tlb’ fetch_temp !

That’s for the bug.

But regarding the fix…

Turning fetch_temp from uint16_t to uin64_t solves my testcase, but I suspect (?) this might not be correct with regard to the computation of exceptions

Turning fetch_temp from uint16_t to uint32_t would better match my rv32 machine, but leaves the bug for rv64 -hé-hé

Or should I consider rewriting mmu_t:: fetch_jump_table() ?

Anyone, any thought ?

All the best meanwhile

aswaterman, 2025-03-27:
Thanks for the bug report, @jmonesti.  Evidently no one has tested the Zcmt extension on an MMIO device...

We should not change `fetch_temp`'s type, as that would affect the rest of instruction fetch, risking the introduction of new bugs.  The answer is to rewrite `fetch_jump_table` to perform multiple accesses.

Please confirm PR #1945 suffices.

jmonesti, 2025-03-28:
Thank you very much for the quick fix.

Feel free to close the issue.

aswaterman, 2025-03-29:
You're welcome.  Thanks again for the bug report.

PR #1945 (2025-03-29): Fix jump-table accesses in MMIO regions
Resolves #1944
commit 51002ff1 (2025-03-29): Fix jump-table accesses in MMIO regions

Resolves #1944
closed 2025-03-29 by aswaterman