riscv-software-src/riscv-isa-sim #1948: Some questions about the inconsistency between the implementation of spike involving the ssdbltrp extension and the privileged architecture

Baixyzz, 2025-04-07:
From Supervisor-Level ISA . Double Trap Control in sstatus Register,we can see

> When a trap is to be taken into S-mode, if the SDT bit is currently 0, it is then set to 1, and the trap is delivered as expected. However, if SDT is already set to 1, then this is an unexpected trap. In the event of an unexpected trap, a double-trap exception trap is delivered into M-mode. 

However, when using spike to simulate the following scenario, I found that the behavior of spike is inconsistent with the above regulations. When sstatus.SDT=0, vsstatus.SDT=1, medeleg[10]=1, hedeleg[10]=0, the ecall instruction is executed in VS-mode. Then I found that spike traps to M-mode and thinks it is a double-trap exception trap.

From the Supervisor-Level ISA description, there shouldn't be a double trap here. Is it my understanding of the ISA description that is wrong?

aswaterman, 2025-04-07:
cc @ved-rivos - any chance you have time to look into this?

Baixyzz, 2025-04-10:
@ved-rivos Sorry to bother you. Can you share your opinion on this issue?

ved-rivos, 2025-04-10:
@Baixyzz - could I request you to test PR #1954

aswaterman, 2025-04-12:
Thanks for the confirmation, @Baixyzz.

closed 2025-04-12 by aswaterman