riscv-software-src/riscv-isa-sim #1956: Errors with VLEN == 32 configuration

gregdavill, 2025-04-11:
## Summary

The riscv vector specification lists an embedded configuration: zve32x, and lists that it has a minimum VLEN of 32. Spike accepts this as options on the isa string, and some vector operation work, But many cause an assertion error.
```
spike: ../riscv/vector_unit.cc:71: T& vectorUnit_t::elt(reg_t, reg_t, bool) [with T = long unsigned int; reg_t = long unsigned int]: Assertion `(VLEN >> 3)/sizeof(T) > 0' failed.
```

It seems that in the vector implementation internal working copies and masks currently create 64bit types, this causes an assertion to trip.

I think this is the underlying issue faced in #1685. Here is a test-case which activates this with vanilla spike.

## Test case

vlen32.S
```asm
.globl _start
.section .text
_start:
    // Enable vector instructions
    li x10, 0x200
    csrs mstatus, x10

    li x11, 2
    vsetvli x11, x0, e32, m1, ta, ma
    vadd.vx v24, v28, x12, v0.t
    ret
```

link.ld
```
OUTPUT_ARCH("riscv")
ENTRY(_start)

SECTIONS
{
    . = 0x80000000;
    .text : { *(.text) }
}
```

Commands to build and run
```bash
$ riscv-none-elf-gcc -nostdlib -nostartfiles -march=rv32imac_zve32x_zicsr -T link.ld vlen32.S -o vlen32.elf
$ spike --isa=rv32imac_zve32x_zvl32b -l --log-commits --instructions=32 vlen32.elf
core   0: 0x00001000 (0x00000297) auipc   t0, 0x0
core   0: 3 0x00001000 (0x00000297) x5  0x00001000
core   0: 0x00001004 (0x02028593) addi    a1, t0, 32
core   0: 3 0x00001004 (0x02028593) x11 0x00001020
core   0: 0x00001008 (0xf1402573) csrr    a0, mhartid
core   0: 3 0x00001008 (0xf1402573) x10 0x00000000
core   0: 0x0000100c (0x0182a283) lw      t0, 24(t0)
core   0: 3 0x0000100c (0x0182a283) x5  0x80000000 mem 0x00001018
core   0: 0x00001010 (0x00028067) jr      t0
core   0: 3 0x00001010 (0x00028067)
core   0: 0x80000000 (0x20000513) li      a0, 512
core   0: 3 0x80000000 (0x20000513) x10 0x00000200
core   0: 0x80000004 (0x30052073) csrs    mstatus, a0
core   0: 3 0x80000004 (0x30052073) c768_mstatus 0x00000200
core   0: 0x80000008 (0x00004589) c.li    a1, 2
core   0: 3 0x80000008 (0x4589) x11 0x00000002
core   0: 0x8000000a (0x0d0075d7) vsetvli a1, zero, e32, m1, ta, ma
core   0: 3 0x8000000a (0x0d0075d7) c8_vstart 0x00000000 x11 0x00000001 c768_mstatus 0x80000600 c3104_vl 0x00000001 c3105_vtype 0x000000d0
core   0: 0x8000000e (0x01c64c57) vadd.vx v24, v28, a2, v0.t
spike: ../riscv/vector_unit.cc:71: T& vectorUnit_t::elt(reg_t, reg_t, bool) [with T = long unsigned int; reg_t = long unsigned int]: Assertion `(VLEN >> 3)/sizeof(T) > 0' failed.
```

If I adjust the isa string to `--isa=rv32imac_zve32x_zvl64b` this test-case passes.

## Troubleshooting

I've dug into the code a little and it appears that for any instruction with a mask, the mask is created using a `VU.elt<uint64_t>` type. This correctly trips the assertion since elements were configured as 32bits.

I've managed to perform a trivial change mostly in v_ext_macros.h, and some vector instructions, with a direct repalcement uint64_t > uint32_t. This fixes my use_case, but obviously breaks the more typical VLEN >= 64. 

I'm happy to rework this into a more general fix and open a PR.

aswaterman, 2025-04-15:
@chihminchao can you look into this one?

chihminchao, 2025-04-15:
The main vector implementation started from v0.7.1  At that moment, there was only completed V extension, no specific upper bound of vlen, and no zveXX extensions.   Besides,  to use 32b VLEN makes vector unit like SIMD and I didn't think there should be configuration setup with this.  For performance concern, I chosed to 64b as the internal type to implement and added the limitation to miminal VLEN=64 in spike.  

I will improve the checking and error message on VLEN with zve32x first
@gregdavill  Any improvement is welcome.

aswaterman, 2025-04-15:
I’m fine with restricting Spike to VLEN >= 64 for now, as long as the error messaging is more user-friendly. We can always support VLEN=32 in future.

gregdavill, 2025-04-15:
I understand the desire to keep this code performant. Especially given that most vector usage is on  XLEN >= 64 cores.

I did go down the path of modifying the v_ext_macros.h to handle both uint32_t and uint64_t. That ends up as a substantial change, adding overhead just for this edge-case.

A simpler approach may be to handle this special case at the assert, with something like this:
```diff
diff --git a/riscv/vector_unit.cc b/riscv/vector_unit.cc
index a5c35f8f..4ee794fe 100644
--- a/riscv/vector_unit.cc
+++ b/riscv/vector_unit.cc
@@ -68,8 +68,8 @@ reg_t vectorUnit_t::vectorUnit_t::set_vl(int rd, int rs1, reg_t reqVL, reg_t new
 
 template<class T> T& vectorUnit_t::elt(reg_t vReg, reg_t n, bool UNUSED is_write) {
   assert(vsew != 0);
-  assert((VLEN >> 3)/sizeof(T) > 0);
-  reg_t elts_per_reg = (VLEN >> 3) / (sizeof(T));
+  assert(((VLEN >> 3)/sizeof(T) > 0) || (VLEN == 32));
+  reg_t elts_per_reg = std::min((VLEN >> 3) / (sizeof(T)), (reg_t)1);
   vReg += n / elts_per_reg;
   n = n % elts_per_reg;
 #ifdef WORDS_BIGENDIAN
```

On a XLEN==VLEN==32 system, when a mask or internal uint64_t elt is created. This calculation used to produce  0, the assert catches is. With this patch the assert is bypassed for VLEN==32, elts_per_reg clamped to 1, and then the upper bits of the mask discarded. 

This patch works for the example I've provided, and some riscv-dv vector instruction streams.

PR #1972 (2025-04-30): Fix mask element accesses under VLEN=32 
This is mostly just a bunch of DRY; the old mask-handling code was copied and pasted a little too much.  But I'd appreciate a careful review, @jerryz123 

Resolves #1956
closed 2025-04-30 by aswaterman