riscv-software-src/riscv-isa-sim #1962: Question about installation

dpetrisko, 2025-04-16:
Hello,

I am trying to reduce installation size for a disk constrained system and I'm curious on what some of the items in the installation are and if they are required. I couldn't find anything in the documentation but please feel free to point me there if I missed something!

- xspike / termios-xspike 
  - what are these used for? If I'm just using regular spike for batch simulations, will I need these?
- spike-log-parser
  - I understand this is used to pretty print --log from spike, but is there a configure option to disable building this feature? I thought I recalled there was one way back but I don't see it
- libcustomext.so
  - Is this only needed if I am using custom extensions defined by spike? 
- libriscv.so
  - In general, could lib/ binary size be reduced by reducing the extensions enabled in spike? For instance, removing vector extension from the library rather than just using the isa string to disable at runtime?

Thank you!

jerryz123, 2025-04-16:
You don't need `xspike` when just running `spike` normally.

`spike-log-parser` is a command line utility for disassembling instruction bits in commit logs... it is not related to the main spike binary.

`libcustomext.so` is an example custom extension. You do not need to install this.

`libriscv.so` is not used by spike-main, since spike-main is statically linked.

Running `strip` on spike got it down to 8 MB for me.

dpetrisko, 2025-04-17:
Thank you, that's all super helpful!

closed 2025-04-17 by dpetrisko