riscv-software-src/riscv-isa-sim #1964: [Bug Report] Incorrect Instruction Fetch in Spike after `ebreak` and `unimp`

LuLuji04, 2025-04-18:
### Description

In my test case, the instruction `lw x10, 0(x19)` at address `0x80000258` is misfetched as `c.fsd fs0, 8(a0)` (a compressed floating-point store instruction) instead of the expected load instruction `lw x10, 0(x19)`. This misfetch causes a `trap_store_access_fault` exception instead of correctly executing the load instruction, which should cause a `trap_load_access_fault`.

[image]
[image]

### Expected Behavior

Spike should correctly fetch and execute the instruction `lw x10, 0(x19)` as a load instruction, which will causes `trap_load_access_fault`.

### What I Did in the Test

Before triggering this bug, I executed the `ebreak` instruction, which caused a `trap_breakpoint`. Upon returning from the breakpoint, due to the address being misaligned, a `c.unimp` instruction was triggered, resulting in a `trap_illegal_instruction` exception. After this, when the program returned again, the observed bug occurred.

### To Reproduce

To reproduce this issue, follow these steps:

1. **Run the test** by executing the `make run` command and observe the output log;
2. **Output disassembly** by using the  `make disasm` command to generate a disassembly file of the binary:

Here is the test code:
[test.zip](https://github.com/user-attachments/files/19814768/test.zip)

### Environment

- Spike branch: master
- Spike commit id: 6268f2ead5595a40af1e27943f42a39846a37c9d

aswaterman, 2025-04-18:
Not a spike bug; your program is overwriting the instruction.  (But thank you for providing a test case so that this was easy to debug.)

For your future reference, this is how I debugged it.  I looked at the initial memory contents where the `lw` instruction is, then ran the simulator until that memory location changed.  I then stepped the simulator by one instruction to get the `pc`, indicating that the previous instruction is the culprit.  That's `sb s5, 24(t0)` instruction, so I checked its base register `t0`, and indeed `24(t0)` is the address of the overwritten instruction.

```
$ spike -d test
(spike) mem 0 80000258
0x10ac01530009a503
(spike) while mem 0 80000258 0x10ac01530009a503
(spike) 
core   0: >>>>  _l158
core   0: 0x0000000080000246 (0x00001997) auipc   s3, 0x1
(spike) reg 0 t0
0x0000000080000240
(spike)

LuLuji04, 2025-04-19:
Thank you very much for your reply. I discovered this issue while performing differential testing between Spike and `XiangShan`, so I will reference this issue in the `XiangShan` repository. The implementation in Spike is indeed correct.

closed 2025-04-19 by aswaterman