riscv-software-src/riscv-isa-sim #1973: Incorrect vmxnor.mm result

gregdavill, 2025-04-30:
# Summary

The changes done in #1972 seem to have introduced a subtle bug in some of the vector operations, specifically ones specifying an inverted boolean result using the bitwise invert operation `~`.

# Details

test case
vmxnor.S
```
.globl _start
.section .text
_start:
    // Enable vector instructions
    li x10, 0x200
    csrs mstatus, x10
    vsetivli x1, 1, e32, m1, ta, ma
    # Load Values
    li x1, 0x05
    vmv.v.x v0, x1
    li x1, 0x01
    vmv.v.x v1, x1

    # Configure vector and execute
    vsetivli x1, 4, e8, m1, ta, ma
    vmxnor.mm v2,v1,v0
    vmxnor.mm v2,v0,v1
    ret
```

```
$ riscv-none-elf-gcc -nostdlib -nostartfiles -march=rv32imac_zve32x_zicsr -T link.ld vset.S -o vset.elf
$ riscv-none-elf-objdump -d vset.elf > vset.lst
$ spike --isa=rv32imac_zve32x_zvl64b -l --log-commits --instructions=32 vset.elf
warning: tohost and fromhost symbols not in ELF; can't communicate with target
core   0: 0x00001000 (0x00000297) auipc   t0, 0x0
core   0: 3 0x00001000 (0x00000297) x5  0x00001000
core   0: 0x00001004 (0x02028593) addi    a1, t0, 32
core   0: 3 0x00001004 (0x02028593) x11 0x00001020
core   0: 0x00001008 (0xf1402573) csrr    a0, mhartid
core   0: 3 0x00001008 (0xf1402573) x10 0x00000000
core   0: 0x0000100c (0x0182a283) lw      t0, 24(t0)
core   0: 3 0x0000100c (0x0182a283) x5  0x80000000 mem 0x00001018
core   0: 0x00001010 (0x00028067) jr      t0
core   0: 3 0x00001010 (0x00028067)
core   0: 0x80000000 (0x20000513) li      a0, 512
core   0: 3 0x80000000 (0x20000513) x10 0x00000200
core   0: 0x80000004 (0x30052073) csrs    mstatus, a0
core   0: 3 0x80000004 (0x30052073) c768_mstatus 0x00000200
core   0: 0x80000008 (0xcd00f0d7) vsetivli ra, 1, e32, m1, ta, ma
core   0: 3 0x80000008 (0xcd00f0d7) x1  0x00000001 c8_vstart 0x00000000 c768_mstatus 0x80000600 c3104_vl 0x00000001 c3105_vtype 0x000000d0
core   0: 0x8000000c (0x00004095) c.li    ra, 5
core   0: 3 0x8000000c (0x4095) x1  0x00000005
core   0: 0x8000000e (0x5e00c057) vmv.v.x v0, ra
core   0: 3 0x8000000e (0x5e00c057) e32 m1 l1 v0  0x0000000000000005 c8_vstart 0x00000000
core   0: 0x80000012 (0x00004085) c.li    ra, 1
core   0: 3 0x80000012 (0x4085) x1  0x00000001
core   0: 0x80000014 (0x5e00c0d7) vmv.v.x v1, ra
core   0: 3 0x80000014 (0x5e00c0d7) e32 m1 l1 v1  0x0000000000000001 c8_vstart 0x00000000
core   0: 0x80000018 (0xcc0270d7) vsetivli ra, 4, e8, m1, ta, ma
core   0: 3 0x80000018 (0xcc0270d7) x1  0x00000004 c8_vstart 0x00000000 c3104_vl 0x00000004 c3105_vtype 0x000000c0
core   0: 0x8000001c (0x7e102157) vmxnor.mm v2, v1, v0
core   0: 3 0x8000001c (0x7e102157) e8 m1 l4 v2  0x000000000000000f c8_vstart 0x00000000
core   0: 0x80000020 (0x7e00a157) vmxnor.mm v2, v0, v1
core   0: 3 0x80000020 (0x7e00a157) e8 m1 l4 v2  0x000000000000000f c8_vstart 0x00000000
core   0: 0x80000024 (0x00008082) ret
core   0: 3 0x80000024 (0x8082)
core   0: exception trap_instruction_access_fault, epc 0x00000004
core   0:           tval 0x00000004
```

The result of `0xf` is incorrect, I would expect `0xb` from a mask operation of `~(0x5 ^ 0x1)` and vl=4

# Debugging

I found the issue against vmxnor.mm

vmxnor_mm.h
```
// vmnxor.mm vd, vs2, vs1
VI_LOOP_MASK(~(vs2 ^ vs1));
```

Note the outer `~`. 
vs1 and vs2 are now bools. The bitwise invert of the boolean result is creating incorrect behviour.

```
true = 1
false = 0
# 32bits shown, but the issue extends
~(true) = ~(1) = 0xFFFF_FFFE = -2 == true
~(false) = ~(0) = 0xFFFF_FFFF = -1 == true
```

Since both of these aren't 0, both resolve to true. So the result from vmxnor_mm is always true for each bit.

jerryz123, 2025-04-30:
Thanks for finding this so quickly. #1974 should cover all the cases.

closed 2025-04-30 by aswaterman

gregdavill, 2025-04-30:
Thanks for fixing this so quickly! My testbench that found this is passing again now.

aswaterman, 2025-04-30:
Cool, thanks for the confirmation.

PR #1974 (2025-04-30): Use boolean not bitwise operators in VI_LOOP_MASK macro
Resolves #1973
Resolves regression introduced by #1972