riscv-software-src/riscv-isa-sim #1976: spike should take exception on illegal overlap of vector macc instructions

antonblanchard, 2025-05-02:
Based on the discussion in https://github.com/riscv/riscv-isa-manual/issues/1789, I believe this should be an illegal overlap because v10 is classed as an input and output register:

```
vwmaccu.vv v10, v11, v1
```

aswaterman, 2025-05-02:
That's mostly right.  This is indeed a reserved instruction, but reserved instructions are not mandatorily illegal.  So Spike's decision not to trap this instruction isn't technically a bug.  Still, I'd prefer Spike be as strict as possible in these cases, to help catch software bugs early.  Over time, I'd like these to be fixed.

aswaterman, 2025-05-05:
Let's categorize and aggregate these issues inside this one ticket.

davidharrishmc, 2026-08-18:
Looks like overlaps should be checked for vwmaccu, vwmacc, vwmaccsu, vwmaccus, vfwmacc, vfwnmacc, vfwmsac, vfwnmsac, and vfwmaccbf16.vv.

It would be helpful to fix this issue if Spike is aiming for Ssstrict support related to certification testing.

aswaterman, 2026-08-18:
Spike doesn't currently claim to support Ssstrict.  It's probably a good thing to do in the long run, but it isn't a priority at the moment.