riscv-software-src/riscv-isa-sim #1979: Why amo instructions require address alignment even when -misaligned option is applied？

13336133, 2025-05-08:
The RISC-V Instruction Set Manual: Volume II Privileged Architecture (Version 20241017)

In 3.6.4. Misaligned Atomicity Granule PMA:

> The misaligned atomicity granule PMA applies only to AMOs
> If a misaligned AMO accesses a region that does not specify a misaligned atomicity granule PMA, or if not
> all accessed bytes lie within the same misaligned atomicity granule, then an exception is raised

Spec said that a misaligned AMO won't raise exception as long as misaligned atomicity granule PMA is applied and all accessed bytes lie within the same misaligned atomicity granule. 

I have test amoadd.d.d on 0xc7, trap_store_access_fault is raised, which violates Spec.
```
li t2, 0xc7
li a5, 0x3
li a6, 0x2
amoadd.d a6, a5, (t2)
```

spike cmd is 
```
spike -d --pc=0x80000000 --misaligned --isa=rv64gc test_amo_misaligned.elf
```

From spike source code:

riscv/mmu.h(179-186)
```
  T amo(reg_t addr, op f) {
    convert_load_traps_to_store_traps({
      store_slow_path(addr, sizeof(T), nullptr, {}, false, true);
      auto lhs = load<T>(addr);
      store<T>(addr, f(lhs));
      return lhs;
    })
  }
```

riscv/mmu.cc(354-360)
```
  if (transformed_addr & (len - 1)) {
    bool gva = access_info.effective_virt;
    if (!is_misaligned_enabled())
      throw trap_store_address_misaligned(gva, transformed_addr, 0, 0);

    if (require_alignment)
      throw trap_store_access_fault(gva, transformed_addr, 0, 0);
```

the require_alignment is always TRUE, regardless of  is_misaligned_enabled

aswaterman, 2025-05-09:
This is not a bug; it is a missing feature. Spike does not yet implement the Zama16b extension. Without that extension, raising the misaligned exception is the correct behavior.

13336133, 2025-05-09:
understood！

closed 2025-05-09 by 13336133