riscv-software-src/riscv-isa-sim #1980: is including mmu.h in all instructions necessary?

farukyld, 2025-05-08:
### Context
I'm maintaining a fork of riscv-isa-sim, where I modify mmu.h to add commit log information (e.g., physical addresses for load/store instructions).

### Problem
Currently, every instruction file includes mmu.h, even those that don't seem to directly depend on it. As a result, any change to mmu.h triggers a full recompilation of all instructions, which takes a few minutes.

### Question
Would it be possible to include mmu.h only where it's actually used, or is there a design reason for the current structure?

I’m not sure if this has been discussed before—apologies if it has.

aswaterman, 2025-05-09:
It would be an extra maintenance burden to explicitly indicate which instructions might depend on the MMU. IMO, the additional complication time is a very minor issue.

farukyld, 2025-05-11:
Thank you for your quick response. I appreciate your explanation. if it is okay, I am closing the issue.

closed 2025-05-11 by farukyld