riscv-software-src/riscv-isa-sim #1981: questions about Sret/Mret

lewislzh, 2025-05-09:
Spike uses the write function of xstatus when executing mret and sret updates.
[sret.h#L46](https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/insns/sret.h#L46)

However, I think the updates to xstatus caused by mret and sret are not explicit CSR write operations.

Consider the following scenario: after supporting the ss/smdbltrp extension, if menvcfg.DTE is disabled, mstatus.sdt and mstatus.spie are both set to 1, mstatus.sie is 0, and an sret instruction is executed in M-mode to return to S-mode. In this case, sie should be restored to the value of spie, and sdt should remain set. However, since Spike invokes the sstatus write function, it incorrectly treats this as an explicit write to sie while sdt is high, resulting in sie not being set to 1 as expected.

I wonder whether it might be more appropriate for the modifications to xstatus made by mret and sret to be handled independently, rather than relying on the explicit CSR write functions.

aswaterman, 2025-05-11:
@ved-rivos is this another one that you should take a look at?

ved-rivos, 2025-05-13:
@aswaterman - I will look into this.

ved-rivos, 2025-06-07:
Sorry, I somehow lost track of this. The cleanest solution may be to add the following logic to SRET and MRET such that the case where xDT and xIE being 1 together does not occur after the controlling xenvcfg.DTE are cleared. 

@aswaterman if you agree I will send a PR with this.

```
SRET:
sstatus.SDT  = sstatus.SDT & menvcfg.DTE
vsstatus.SDT = hstatus.SPV ? vsstatus.SDT & menvcfg.DTE & henvcfg.DTE : vstatus.SDT

MRET:
sstatus.SDT  = mstatus.MPP == M  ? sstatus.SDT : sstatus.SDT & menvcfg.DTE
vsstatus.SDT = mstatus.MPP == M  ? vsstatus.SDT : 
               mstatus.MPV == 0  ? vstatus.SDT & menvcfg.DTE : 
                                   vstatus.SDT & menvcfg.DTE & henvcfg.DTE

```

lewislzh, 2025-06-09:
I’d like to ask whether sdt is still being modified through the write function of sstatus/vsstatus here. However, according to the write function of sstatus/vsstatus, the sdt bit cannot be modified when menvcfg.DTE is disabled.[csrs.cc#517](https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/csrs.cc#517)

aswaterman, 2025-06-09:
@ved-rivos Are there any analogous situations with `envcfg` already?  If so, which behavior does Spike prefer: do we mask on reads but keep the underlying state bit intact, or do we force the underlying state bit to 0 to avoid the invalid combination?

As long as we are being consistent, I agree with your proposal; please submit a PR.

ved-rivos, 2025-06-09:
@aswaterman we usually mask on read. However, the condition here is that menvcfg does not affect read with M-mode privilege. What we need is a model for the CSR field modulation rule where the SDT bit of the CSR is affected by menvcfg when read at privilege less than M. I will make a PR.

@lewislzh - it would not matter because xIE can be set to 1 if xDT is already 0 **or is being set to 0 by the same write**.

lewislzh, 2025-06-10:
> [@lewislzh](https://github.com/lewislzh) - it would not matter because xIE can be set to 1 if xDT is already 0 **or is being set to 0 by the same write**.

@ved-rivos What I mean is that since menvcfg.DTE is already disabled, if you're writing xDT to 0 in the same write operation, that action will be ignored — therefore, xIE still cannot be set to 1. 
If you write xDT to 0 in a separate write operation, that won’t work either.
The Spike implementation also behaves this way.[csrs.cc#L516-L519](https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/csrs.cc#L516-L519)
I believe the behavior of xret in restoring certain fields of sstatus should not be treated as a write operation, but rather as an update operation.