riscv-software-src/riscv-isa-sim #1994: Shang-Mi 4 vector instruction endianness

kfahrn, 2025-05-16:
The Shang-Mi 4 vector instructions as implemented in [vsm4k_vi](https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/insns/vsm4k_vi.h), [vsm4r_vs](https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/insns/vsm4r_vs.h) and [vsm4r_vv](https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/insns/vsm4r_vv.h) all use little endian ordering in their implementations. However, in the [ratified specifications](https://drive.google.com/file/d/1uviu1nH-tScFfgrovvFCrj7Omv8tFtkp/view) on pages 564-569 where the instructions are defined, they appear to be using big endian ordering. This leads to calculations not matching as the four words that are being processed are different both in endianness and ordering.

Below are excerpts from the SM4 key expansion of Spike and then the sail code from the specifications. It is clearly visible that the word ordering is reversed as Spike initiates the operands by extracting them in little endian with `EXTRACT_EGU32x4_WORDS_LE()`.

The Spike implementation in [vsm4k_vi](https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/insns/vsm4k_vi.h):
```cpp
VI_ZVK_VD_VS2_ZIMM5_EGU32x4_NOVM_LOOP(
  {},
  // The following statements will be executed before the first execution
  // of the loop, and only if the loop is going to be entered.
  // We cannot use a block ( { ... } ) since we want the 'round' variable
  // declared and defined here  here to be visible in the loop block.
  // Only consider the bottom 3 bits of the immediate, ensuring that
  // 'round' is in the valid range [0, 7].
  const reg_t round = zimm5 & 0x7;,
  // Per Element Group body.
  {
    // {rk0, rk1, rk2, rk3} <- vs2
    EXTRACT_EGU32x4_WORDS_LE(vs2, rk0, rk1, rk2, rk3);

    uint32_t B = rk1 ^ rk2 ^ rk3 ^ zvksed_ck[4 * round];
    uint32_t S = ZVKSED_SUB_BYTES(B);
    uint32_t rk4 = ZVKSED_ROUND_KEY(rk0, S);

    B = rk2 ^ rk3 ^ rk4 ^ zvksed_ck[4 * round + 1];
    S = ZVKSED_SUB_BYTES(B);
    uint32_t rk5 = ZVKSED_ROUND_KEY(rk1, S);

    B = rk3 ^ rk4 ^ rk5 ^ zvksed_ck[4 * round + 2];
    S = ZVKSED_SUB_BYTES(B);
    uint32_t rk6 = ZVKSED_ROUND_KEY(rk2, S);

    B = rk4 ^ rk5 ^ rk6 ^ zvksed_ck[4 * round + 3];
    S = ZVKSED_SUB_BYTES(B);
    uint32_t rk7 = ZVKSED_ROUND_KEY(rk3, S);

    // Update the destination register.
    SET_EGU32x4_LE(vd, rk4, rk5, rk6, rk7);
  }
);
```

The vsm4k_vi specification from the [ratified specifications](https://drive.google.com/file/d/1uviu1nH-tScFfgrovvFCrj7Omv8tFtkp/view) on pages 564-565:
```text
{
eg_len = (vl/EGS)
eg_start = (vstart/EGS)
let B : bits(32) = 0;
let S : bits(32) = 0;
let rk4 : bits(32) = 0;
let rk5 : bits(32) = 0;
let rk6 : bits(32) = 0;
let rk7 : bits(32) = 0;
let rnd : bits(3) = uimm[2:0]; // Lower 3 bits
foreach (i from eg_start to eg_len-1) {
let (rk3 @ rk2 @ rk1 @ rk0) : bits(128) = get_velem(vs2, 128, i);
B = rk1 ^ rk2 ^ rk3 ^ ck(4 * rnd);
S = sm4_subword(B);
rk4 = ROUND_KEY(rk0, S);
B = rk2 ^ rk3 ^ rk4 ^ ck(4 * rnd + 1);
S = sm4_subword(B);
rk5 = ROUND_KEY(rk1, S);
B = rk3 ^ rk4 ^ rk5 ^ ck(4 * rnd + 2);
S = sm4_subword(B);
rk6 = ROUND_KEY(rk2, S);
B = rk4 ^ rk5 ^ rk6 ^ ck(4 * rnd + 3);
S = sm4_subword(B);
rk7 = ROUND_KEY(rk3, S);
// Update the destination register.
set_velem(vd, EGW=128, i, (rk7 @ rk6 @ rk5 @ rk4));
}
RETIRE_SUCCESS
}
```

This appears to not just be a difference in naming order, but also calculation order, since in both implementations the fourth step of the key expansion (rk7 and x7 respectively) are being calculated last but are written back to different ends of the result vector vd.

The same issue also exists for the other two SM4 instructions, [vsm4r_vs](https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/insns/vsm4r_vs.h) and [vsm4r_vv](https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/insns/vsm4r_vv.h).

aswaterman, 2025-05-16:
cc @nibrunieAtSi5

nibrunieAtSi5, 2025-05-16:
We might need to involve Ken Dockser here.

Looking at the vector crypto code samples, I am surprised to see `vrgather` to perform element swap and that no one complained: https://github.com/riscv/riscv-crypto/blob/ef1eeb9074e57d4e08d77d3db43dc345e354bc7f/doc/vector/code-samples/zvksed.s#L176-L184 (key round) and https://github.com/riscv/riscv-crypto/blob/ef1eeb9074e57d4e08d77d3db43dc345e354bc7f/doc/vector/code-samples/zvksed.s#L200 (encryption/decryption)

nibrunieAtSi5, 2025-05-16:
trying to make sense of everything, spike currently uses 
1. to extract the elements

```
    // {rk0, rk1, rk2, rk3} <- vs2
    EXTRACT_EGU32x4_WORDS_LE(vs2, rk0, rk1, rk2, rk3);
```

https://github.com/riscv-software-src/riscv-isa-sim/blob/14cad996bfd0def62d57dfc4ecf51e7e1d8df3f5/riscv/zvk_ext_macros.h#L847-L852
```
#define EXTRACT_EGU32x4_WORDS_LE(X, W0, W1, W2, W3) \
  uint32_t W0 = (X)[0]; \
  uint32_t W1 = (X)[1]; \
  uint32_t W2 = (X)[2]; \
  uint32_t W3 = (X)[3]; \
  (void)(0)
```

2. to build the results

```
    // Update the destination register.
    SET_EGU32x4_LE(vd, rk4, rk5, rk6, rk7);
```

https://github.com/riscv-software-src/riscv-isa-sim/blob/14cad996bfd0def62d57dfc4ecf51e7e1d8df3f5/riscv/zvk_ext_macros.h#L858C1-L864C14
```
#define SET_EGU32x4_LE(X, W0, W1, W2, W3) \
  do { \
    (X)[0] = (W0); \
    (X)[1] = (W1); \
    (X)[2] = (W2); \
    (X)[3] = (W3); \
  } while (0)
```

which expands to `(X)[0] = rk4`, ... up to `(X)[3] = rk7`, right ?

and the ISA spec "SAIL" uses:

```
let (rk3 @ rk2 @ rk1 @ rk0) : bits(128) = get_velem(vs2, 128, i);

```

and

```
set_velem(vd, EGW=128, i, (rk7 @ rk6 @ rk5 @ rk4));

```

But isn't SAIL syntax
https://alasdair.github.io/manual.html#_accessing_and_updating_vectors
```
The @ operator is used to concatenate bitvectors, for example:

let x = 0xFFFF;
let y = 0x0000;
assert(x @ y == 0xFFFF_0000);
```
expecting the same order, in `rk7 @ rk6 @ rk5 @ rk4` `rk7` is the most significant element which appears on the highest bits in the vector register ?

(I am not very familiar with the SAIL syntax so I might be missing something obvious)

(cc @kdockser)

jordancarlin, 2025-05-16:
@nadime15 has been working on the actual Sail implementation of these instructions (https://github.com/riscv/sail-riscv/pull/848), which might be worth comparing. What is in the spec right now is just a Sail-like pseudocode.

nibrunieAtSi5, 2025-05-16:
Thank you @jordancarlin , I was going to link the PR, it looks like the syntax for vector extraction and building is indeed quite different.

nadime15, 2025-05-17:
As @jordancarlin pointed out, the Sail code in the spec is more like Sail-like pseudocode. Some parts are not valid Sail and is a mix between Sail and Verilog. For example, the following line is not correct Sail:

```let (rk3 @ rk2 @ rk1 @ rk0) : bits(128) = get_velem(vs2, 128, i);```

I think the Author had something like this in mind (in verilog):

` assign {rk3, rk2, rk1, rk0} = get_velem(vs2, 128, i);`

which would be handled in little endian order.

But I agree, it kind of gives the impression that things are being processed in big endian. There are several code snippets where the Sail-like pseudocode in the spec doesn’t fully align with real, working Sail code and mixes Sail and Verilog.  

The Sail PR for Zvksed [#848](https://github.com/riscv/sail-riscv/pull/848) addresses this, and it matches what Spike is doing, processing elements in little endian.

nibrunieAtSi5, 2025-05-17:
Looking at implementations in the wild (unrelated to spike per say). It seems OpenSSL relies on strided memory accesses to swap element orders.

For example here when setting the decrypt key:
```
    @{[vsse32_v $vk7, $keys, $stride]} # rk[31:28]
    addi $keys, $keys, 16
    @{[vsse32_v $vk6, $keys, $stride]} # rk[27:24]
    addi $keys, $keys, 16
```
https://github.com/openssl/openssl/blob/10bd6fa8ca93b4cf53f005f110c827ed923c89a4/crypto/sm4/asm/sm4-riscv64-zvksed.pl#L154-L168

Or here (both a byte ordering reversal in elements + strided store) when encrypting:

```
    # Save the ciphertext (in reverse element order)
    @{[vrev8_v $vdata, $vdata]}
    li $stride, -4
    addi $out, $out, 12
    @{[vsse32_v $vdata, $out, $stride]}

```
https://github.com/openssl/openssl/blob/10bd6fa8ca93b4cf53f005f110c827ed923c89a4/crypto/sm4/asm/sm4-riscv64-zvksed.pl#L221-L227

kdockser, 2025-05-22:
It has been a while since I wrote that _Sail-like pseudo code_, but I think that @nibrunieAtSi5 and @nadime15 are correct that it was my intention to have the little endian behavior of `assign {rk3, rk2, rk1, rk0} = get_velem(vs2, 128, i);`

I think the best approach will be to update the SM4 definitions in the unpriv spec with the actual SAIL code, once it has been tested.

nibrunieAtSi5, 2025-06-28:
Looking at this into a bit more details, I am not sure there was an obvious better solution when the instructions were defined.

The key schedule is not element-swapped for the forward encryption, it is only element-swapped for the backward decryption. So with a single instruction to expand the key schedule, either forward or backward needs to be element reversed. 
The key schedule being re-used during a full buffer/session encryption / decryption, the cost of the element reversed (even if the `vrgather` / strides-vmem are slow) is amortized.

The encryption/decryption is always followed by an element swap before the result block can be used (e.g. stored in memory), but no input element swap is required. This is true for both encryption and decryption. Here again, if we swap the instruction output element order, I believe this would imply having to swap the input to the first encryption/decryption round instruction. So basically we would exchange a final swap by an initial swap. No savings here either. We could introduce a new instruction, e.g. `vsm4rz` (initial round) or `vsm4rf` (final round) to perform round + element swap and save the extra instruction.

_I will continue digging._