riscv-software-src/riscv-isa-sim #2003: Incorrect dest vector elements access for vector widening insns

HaochenGui, 2025-06-10:
For vector widening insns (e.g. vwmul), they invokes VI_VV_LOOP_WIDEN then VV_PARAMS to set up vd, vs1 and vs2.  The EEW of vd should be doubled. But the codes still use SEW, which may cause accessing wrong physical vector register.  The untouched physical vector register (mark out by mask) might be considered as modified in state as the rd accessing is writable.

I made a patch for this problem.  Could someone inform me how to submit the patch for review? Thanks.

aswaterman, 2025-06-11:
@chihminchao I would be very surprised if this had gone unnoticed. I'll leave it to you to respond to.

HaochenGui, 2025-06-12:
I paste my patch here.  

From 2738bd4539a4b5b085be63a7d454a409efb17281 Mon Sep 17 00:00:00 2001
From: Haochen Gui <haochen.gui@riscv-computing.com>
Date: Mon, 9 Jun 2025 09:13:47 +0000
Subject: [PATCH] Correct EEW of dest vector for vector widen insns

* riscv/v_ext_macros.h (VWV_PARAMS): Define.
  (VI_VV_LOOP_WIDEN): Replace VV_PARAMS with VWV_PARAMS.
---
 riscv/v_ext_macros.h | 11 ++++++++---
 1 file changed, 8 insertions(+), 3 deletions(-)

diff --git a/riscv/v_ext_macros.h b/riscv/v_ext_macros.h
index 1e33232d..f0a4fc84 100644
--- a/riscv/v_ext_macros.h
+++ b/riscv/v_ext_macros.h
@@ -329,6 +329,11 @@ static inline bool is_overlapped_widen(const int astart, int asize,
   type_sew_t<x>::type vs1 = P.VU.elt<type_sew_t<x>::type>(rs1_num, i); \
   type_sew_t<x>::type UNUSED vs2 = P.VU.elt<type_sew_t<x>::type>(rs2_num, i);

+#define VWV_PARAMS(x) \
+  type_sew_t<x*2>::type UNUSED &vd = P.VU.elt<type_sew_t<x*2>::type>(rd_num, i, true); \
+  type_sew_t<x>::type vs1 = P.VU.elt<type_sew_t<x>::type>(rs1_num, i); \
+  type_sew_t<x>::type UNUSED vs2 = P.VU.elt<type_sew_t<x>::type>(rs2_num, i);
+
 #define VX_PARAMS(x) \
   type_sew_t<x>::type UNUSED &vd = P.VU.elt<type_sew_t<x>::type>(rd_num, i, true); \
   type_sew_t<x>::type rs1 = (type_sew_t<x>::type)RS1; \
@@ -903,13 +908,13 @@ static inline bool is_overlapped_widen(const int astart, int asize,
 #define VI_VV_LOOP_WIDEN(BODY) \
   VI_LOOP_BASE \
   if (sew == e8) { \
-    VV_PARAMS(e8); \
+    VWV_PARAMS(e8); \
     BODY; \
   } else if (sew == e16) { \
-    VV_PARAMS(e16); \
+    VWV_PARAMS(e16); \
     BODY; \
   } else if (sew == e32) { \
-    VV_PARAMS(e32); \
+    VWV_PARAMS(e32); \
     BODY; \
   } \
   VI_LOOP_END
--
2.43.0

nibrunieAtSi5, 2025-06-28:
@HaochenGui , do you have a test case where this fails ?
As far as I can tell, the widening ops rely on `VI_WIDE_OP_AND_ASSIGN`
e.g. https://github.com/riscv-software-src/riscv-isa-sim/blob/14670706c1ef819f2d2b7e656eec3241b2ac80b8/riscv/insns/vwmul_vv.h

```
// vwmul.vv vd, vs2, vs1
VI_CHECK_DSS(true);
VI_VV_LOOP_WIDEN
({
  VI_WIDE_OP_AND_ASSIGN(vs2, vs1, 0, *, +, int);
})
```

 which performs the proper `2 * SEW` assignment of elements:

```
#define VI_WIDE_OP_AND_ASSIGN(var0, var1, var2, op0, op1, sign) \
  switch (P.VU.vsew) { \
  case e8: { \
    sign##16_t UNUSED vd_w = P.VU.elt<sign##16_t>(rd_num, i); \
    P.VU.elt<uint16_t>(rd_num, i, true) = \
      op1((sign##16_t)(sign##8_t)var0 op0 (sign##16_t)(sign##8_t)var1) + var2; \
    } \
    break; \
(...)

```

from https://github.com/riscv-software-src/riscv-isa-sim/blob/14670706c1ef819f2d2b7e656eec3241b2ac80b8/riscv/v_ext_macros.h#L931-L951

HaochenGui, 2025-06-30:
> * What was the value of `v9` before the execution of the `vwmul.vv` ?
The original value of v9 is 0x00000000000000000000000000000000. It's untouched in the whole process of vwmul.
 
> * What an instruction reading `v9` and `v10` sees after the `vwmul.vv` at PC `0x80000038` ?
>   (I am trying to understand if this is just a trace issue [1] or if there is actually a wrong architectural write to an invalid value going on.x
It's a trace problem.  In  VV_PARAMS, the reference of wrong elements of vd is set up.  But it's never be used later (by write).  For v9, the content of register is correct, but it's marked as written and shown in the trace. That's problem. 

` #define VV_PARAMS(x) \
  type_sew_t<x>::type UNUSED &vd = P.VU.elt<type_sew_t<x>::type>(rd_num, i, true); \`

Thanks. @nibrunie