riscv-software-src/riscv-isa-sim #2011: Does vadd.vv require zero vstart?

HaochenGui, 2025-06-19:
The vadd.vv invokes VI_VV_LOOP ->VI_LOOP_BASE->VI_GENERAL_LOOP_BASE->require_vector(true).  It means that vadd.vv requires a zero vstart value.  It raises illegal instruction if the vstart is non-zero.

According to macro require_vector, it requires a zero vstart if alu is true as P.VU.vstart_alu is always false.
```
 #define require_vector(alu) \
  do { \
    require_vector_vs; \
    require(!P.VU.vill); \
    if (alu && !P.VU.vstart_alu) \
      require(P.VU.vstart->read() == 0); \
    WRITE_VSTATUS; \
    dirty_vs_state; \
  } while (0);
```

The problem occurs on lots of vector instructions.  But the ISA only requires zero vstart for vector reduction, vcpop, vfirst, vmsbf, vmsif, vmsof  and viota.

chihminchao, 2025-06-19:
It provides an implementation choice for non-zero vstart with ALU instruction.  I think the original goal for vstart is to record the position when exception happens.  However, exception couldn't be triggered in the middle of ALU operations.   To make it adjustable only to load/store instructions may simplify u-arch.

aswaterman, 2025-06-19:
Because requiring vstart=0 for ALU ops simplifies the uarch a bit, and has little downside, it is a popular implementation choice (which is allowed by the ISA).  Spike matches the common implementation choice.

HaochenGui, 2025-06-20:
Got it. Thanks so much for your comments. Close this issue.

closed 2025-06-20 by HaochenGui