riscv-software-src/riscv-isa-sim #2012: Why VU accessing supervisor CSRs causes vs insn exception?

xuchengbei, 2025-06-20:
Why VU mode accessing supervisor CSRs like satp (really vsatp) causes a virtual-instruction exception instead of an illegal-instruction exception like when U mode accesses satp? Is there any reason it must be handled by hypervisor in HS-mode instead of guest kernel in VS-mode?

```
void csr_t::verify_permissions(insn_t insn, bool write) const {
  // Check permissions. Raise virtual-instruction exception if V=1,
  // privileges are insufficient, and the CSR belongs to supervisor or
  // hypervisor. Raise illegal-instruction exception otherwise.
  unsigned priv = state->prv == PRV_S && !state->v ? PRV_HS : state->prv;

  if ((csr_priv == PRV_S && !proc->extension_enabled('S')) ||
      (csr_priv == PRV_HS && !proc->extension_enabled('H')))
    throw trap_illegal_instruction(insn.bits());

  if (write && csr_read_only)
    throw trap_illegal_instruction(insn.bits());
  if (priv < csr_priv) {
    if (state->v && csr_priv <= PRV_HS)
      throw trap_virtual_instruction(insn.bits());
    throw trap_illegal_instruction(insn.bits());
  }
}
```

radimkrcmar, 2025-06-20:
The RISC-V manual specifies this behavior.
Authors of the H extension might explain why.  (I assume it has to do with the expected configurations of medeleg/hedeleg -- illegal-instruction exception likely wouldn't get delegated directly to VS-mode anyway.)

gfavor, 2025-06-20:
I believe this is to support nested virtualization/hypervisors.  Past that limited answer, you'll probably need to ask the software people in the Hypervisor SIG.

aswaterman, 2025-06-22:
If it has been clearly determined that there is an actual Spike bug, please file a new issue.