riscv-software-src/riscv-isa-sim #2026: Incorrect trap behavior when vd and vs2 overlap

jacassidy, 2025-07-02:
Running the following assembly on spike causes a trap, while the same code on sail sucessfully completes:

the spec states:

> For vector indexed segment loads, the destination vector register groups cannot overlap the source vector register group (specified by vs2), else the instruction encoding is reserved.

but this constraint is not mentioned for non-segment loads as is tested below.

```# Testcase cp_vd (Test destination vd = v1)
li x27, 1                                         # Load desired vl value
vsetvli x0, x27, e16, m1, tu, mu
la x4, vd_random_base_001                         # Load address of desired value
vle16.v v1, (x4)                                  # Load desired value from memory into v1
la x5, vector_ls_random_base                      # Load address pointer integer register for loads
la x1, vs2_random_base_001                        # Load address of desired value
vle16.v v1, (x1)                                  # Load desired value from memory into v1
csrr x19, vtype                                   # save vtype register for after load
csrr x9, vl                                       # save vl register for after load
vsetvl x10, x0, x19                               # set vl to vlmax
add x10, x10, x10                                 # save vlmax * 2
vsetvli x0, x9, e8, m1, ta, ma                    # setting sew to vs2 eew
vrem.vx v1, v1, x10                               # ensure all values are within (-2*vlmax, 2*vlmax)
vand.vi v1, v1, -2                                # sew-aligning elements
vsetvl x0, x9, x19                                # restore vl and vtype setting
vluxei8.v v1, (x5), v1
RVTEST_SIGUPD_V(x3, x6, 1, 16,  v1)               # stores v1 (sew = 16, AVL = 1) in signature with base (x3) and helper (x6) register```

Spike:

```core   0: 0x80000200 (0x008df057) vsetvli zero, s11, e16, m1, tu, mu
(spike) 
core   0: 0x80000204 (0x00004217) auipc   tp, 0x4
(spike) 
core   0: 0x80000208 (0xe0820213) addi    tp, tp, -504
(spike) 
core   0: 0x8000020c (0x02025087) vle16.v v1, (tp)
(spike) 
core   0: 0x80000210 (0x00006297) auipc   t0, 0x6
(spike) 
core   0: 0x80000214 (0x6b028293) addi    t0, t0, 1712
(spike) 
core   0: 0x80000218 (0x00008097) auipc   ra, 0x8
(spike) 
core   0: 0x8000021c (0x6ac08093) addi    ra, ra, 1708
(spike) 
core   0: 0x80000220 (0x0200d087) vle16.v v1, (ra)
(spike) 
core   0: 0x80000224 (0xc21029f3) csrr    s3, vtype
(spike) 
core   0: 0x80000228 (0xc20024f3) csrr    s1, vl
(spike) 
core   0: 0x8000022c (0x81307557) vsetvl  a0, zero, s3
(spike) 
core   0: 0x80000230 (0x00a50533) add     a0, a0, a0
(spike) 
core   0: 0x80000234 (0x0c04f057) vsetvli zero, s1, e8, m1, ta, ma
(spike) 
core   0: 0x80000238 (0x8e1560d7) vrem.vx v1, v1, a0
(spike) 
core   0: 0x8000023c (0x261f30d7) vand.vi v1, v1, -2
(spike) 
core   0: 0x80000240 (0x8134f057) vsetvl  zero, s1, s3
(spike) 
core   0: 0x80000244 (0x06128087) vluxei8.v v1, (t0), v1
core   0: exception trap_illegal_instruction, epc 0x80000244
core   0:           tval 0x06128087
(spike) 
core   0: exception trap_instruction_access_fault, epc 0x00000000
core   0:           tval 0x00000000```

Sail:

```
[150] [M]: 0x80000240 (0x8134F057) vsetvl x0, x9, x19
CSR mstatus (0x300) <- 0x0000000080000600
CSR mstatush (0x310) <- 0x0000000000000000
CSR vstart (0x008) <- 0x0000000000000000
CSR vtype (0xC21) <- 0x0000000000000008
CSR vl (0xC20) <- 0x0000000000000001
CSR vstart (0x008) <- 0x0000000000000000

mem[X,0x080000244] -> 0x8087
mem[X,0x080000246] -> 0x0612
[151] [M]: 0x80000244 (0x06128087) vluxei8.v v1, (x5), v1
CSR mstatus (0x300) <- 0x0000000080000600
CSR mstatush (0x310) <- 0x0000000000000000
CSR vstart (0x008) <- 0x0000000000000000
mem[R,0x0800068F2] -> 0x4B2D
CSR mstatus (0x300) <- 0x0000000080000600
CSR mstatush (0x310) <- 0x0000000000000000```

aswaterman, 2025-07-03:
Thanks for the bug report, @jacassidy.  I agree, nothing in the spec seems to justify disallowing this register overlap.  I'll dig into it.

aswaterman, 2025-07-03:
Sorry @jacassidy, I misinterpreted the test case.  I think Spike is correct in this case.

`vluxei8.v v1, (x5), v1` with SEW=16 is a widening instruction (it has source EEW 8 and destination EEW 16).  So, overlap is only allowed if the source EMUL is at least 1 (which it isn't--it's 1/2).  To make this instruction valid, you'd need to either avoid overlap, or increase LMUL and make the overlap in the high part, e.g. `vluxei8.v v0, (x5), v1` with SEW=16 and LMUL=2.

Can you refile this a bug against the SAIL model, and also against whatever test generator you're using?

jacassidy, 2025-07-03:
Thank you for the catch! Ill make sure to make a report to both.

closed 2025-07-03 by jacassidy