riscv-software-src/riscv-isa-sim #2064: Does the mnret instruction require setting mnstatus.mnpp to 0?

lipeng-77, 2025-09-03:
https://github.com/riscv-software-src/riscv-isa-sim/blob/9c190a07c6838f6392bafa4ad83acea462c7f759/riscv/insns/mnret.h#L12

From my understanding, mnret should be similar to mret. Since mret sets mstatus.mpp to 0 (if U-mode is implemented), should mnret also include the action of setting mnstatus.mnpp to 0?

https://github.com/riscv-software-src/riscv-isa-sim/blob/9c190a07c6838f6392bafa4ad83acea462c7f759/riscv/insns/mret.h#L10

aswaterman, 2025-09-03:
No, the spec doesn't say that MNRET changes MNPP.  This is a minor divergence from the behavior of MRET, but it doesn't really matter, since MRET's behavior in this respect isn't important for correct software's functionality anyway.

lipeng-77, 2025-09-03:
Thank you for the clarification!

closed 2025-09-03 by lipeng-77