riscv-software-src/riscv-isa-sim #2080: RFC: Integrate llvm-snippy into  riscv-isa-sim's test suit

arrv-sc, 2025-09-10:
# RFC: Integrate llvm-snippy into riscv-isa-sim's test suit

Hi! I want to propose an idea for improving Spike's testing. I'm the developer of [llvm-snippy](https://github.com/syntacore/snippy), a tool that generates random RISC-V code snippets for testing purposes. It supports most of the RISC-V standard extensions and can create interesting control flow with loops and other branches. We've already used it to find and fix several bugs in Spike and even some in sail-riscv (e.g. https://github.com/riscv-software-src/riscv-isa-sim/pull/1995, https://github.com/riscv-software-src/riscv-isa-sim/pull/1992 and https://github.com/riscv/sail-riscv/pull/1090). I think it could be valuable for this project.

### What is llvm-snippy?
llvm-snippy is an open-source llvm-based tool that generates random but valid RISC-V code snippets. It allows you to specify how many instructions to generate and probabilities of generating each of these instructions. This makes it great for testing a wide range of scenarios, from simple arithmetic instruction sequences to complex control flows, floating-point arithmetic with different rounding modes and atomic memory accesses into different memory regions. One of the snippy's best features in my opinion is that it reuses llvm's backend and instruction descriptions, which means that every instruction supported by LLVM and clang (Which is most of the existing RISC-V extensions) is automatically supported in llvm-snippy. That allows snippy not to worry about generating invalid opcodes.

We've been using it extensively at my company, and it's helped us catch a lot of issues that might have gone unnoticed otherwise. I think it could be a great addition to Spike's test suite.

### Why llvm-snippy?

Random instruction testing is a common approach and many cores/simulators use it. For example [sail-riscv](https://github.com/riscv/sail-riscv) uses [riscv-vector-tests](https://github.com/chipsalliance/riscv-vector-tests) in CI. llvm-snippy, however is a much more capable generator. While riscv-vector-tests (and many others) generate single-instruction tests, llvm-snippy can generate many instructions from different extensions in the same snippet. That allows to cover more complex test cases. For example snippet where result of instruction from V extension is a source operand of double-precision operation from D extension. Or where atomics interleave with strided loads/stores to the same memory, etc.

I will not describe all the features snippy has here. However we recently had [2.0 release](https://github.com/syntacore/snippy/releases/tag/snippy-2.0) with extensive documentation and examples that you can check out.

### The Proposal
I'd like to propose integrating llvm-snippy into Spike's regular testing process. The idea is to generate random code for different RISC-V extensions with llvm-snippy and run them on Spike. This alone would check that spike does not fail or trap on said instructions (if it shouldn't). But we can go further and execute these tests on other simulators like sail-riscv or qemu-riscv. By comparing the results (i.e. final register states, memory dumps or instruction traces), we could identify inconsistencies or bugs in the simulators.

This would allow us to:
- Test a wider variety of instruction combinations and control flows that would be impossible to test by-hand.
- Catch bugs that might not be found with existing test cases.
- Cross-validate Spike against other simulators on said code snippets to ensure accuracy.

### Next Steps
I'm happy to implement this testing myself (or with the help of my team). Before diving in, I want to get any feedback, concerns or questions you might have. Are you open to the idea of adding this kind of testing?

Also, if you'd like, I could give a quick presentation or demo (either in comment/text/mail or even videocall format) of llvm-snippy to explain it in more detail. Just let me know if that's something you'd be interested in.

Thanks for considering this! I'm looking forward to hearing your thoughts.

@aswaterman @jerryz123

aswaterman, 2025-09-24:
> We can provide basic sanity tests and check that riscv-isa-sim does not fail on valid code (as I described above).

I prefer this simple approach, at least for the time being.  It provides the most bang for the buck.

Would the idea be that we run snippy hundreds of times offline, then upload the compiled ELFs in a tarball artifact, like we currently do for the ci-tests? https://github.com/riscv-software-src/riscv-isa-sim/blob/master/ci-tests/create-ci-binary-tarball

This approach makes CI faster, and keeping the tests constant prevents nondeterministic CI failures.  In this model, we'd regenerate the tests every few months, as Spike adds new ISA features and/or snippy improves.

arrv-sc, 2025-09-25:
@aswaterman This approach is certainly a valid one. However I have a concern about tarball size. Snippy is capable of generating quite large snippets (hundreds of thousands of instructions) because of that tarball can become fat. Of course this instruction count is configurable and we can come up with numbers that are big enough to have a good probability to catching bugs but not too big so working with tarball slows down. I want to point out that snippy has `--seed`  option and any number of launches with the same seed, config and same snippy version will produce the same snippet. This allows us to still make CI deterministic if we decide to run snippy every time and lock seeds. Time required to generate snippet (even big one) is infinitely lower than execution time of the generated snippet. So I don't think re-generating snippets will affect CI runtime. Overall I don't have strong opinion on this. I think both approaches are fine. Just wanted to mention other options. Which one do you think is best?

arrv-sc, 2025-09-26:
Thanks! I'll create a testing prototype and make a PR soon